unitreerobotics/unitree_guide #1: 在WSL2里Gazebo使用GPU时步态错乱

fan-ziqi, 2023-03-04:
## 环境：
* Windows10
* WSL2最新版本内核
* Ubuntu20.04最新版本
* NVIDIA 3070 Laptop 8G
* 显卡驱动最新版
* ROS-noetic
* Gazebo11

## 问题
最开始在WSL2里使用gazebo仿真，此时步态正常，但发现帧率过低，故去Microsoft官网升级了WSLG，并同时更新了WSL2的内核与显卡驱动，升级前使用LLVM，如下图所示
[image]
升级后如下图，可以检测到独立显卡：
[image]
但此时步态错乱，无法正常行走。
随后将同一份代码拷贝到VMware中同样环境下编译，步态正常，经查看VMware也是用LLVM，**遂怀疑此BUG与独立显卡有关**，下图为Vmware中的显卡信息
[image]

xiaoliangstd, 2023-03-04:
您好，由于没有在Windows环境中测试过unitree_guide, 所以暂时没能分析出这个bug的原因，抱歉！但这边听你描述说 会出现步态错乱，我猜测是不是include/common/timeMarker.h文件中的有些函数在您的环境中不能正常使用？此导致了WaveGenerator.cpp中calcWave()计算步态时序的结果异常。

fan-ziqi, 2023-03-04:
> 您好，由于没有在Windows环境中测试过unitree_guide, 所以暂时没能分析出这个bug的原因，抱歉！但这边听你描述说 会出现步态错乱，我猜测是不是include/common/timeMarker.h文件中的有些函数在您的环境中不能正常使用？此导致了WaveGenerator.cpp中calcWave()计算步态时序的结果异常。

运行环境是WIndows的Linux子系统（WSL2），环境就是原生Linux，唯一的区别就是显卡。能否留个联系方式，我把bug视频发给您看一下

xiaoliangstd, 2023-03-04:
> > 您好，由于没有在Windows环境中测试过unitree_guide, 所以暂时没能分析出这个bug的原因，抱歉！但这边听你描述说 会出现步态错乱，我猜测是不是include/common/timeMarker.h文件中的有些函数在您的环境中不能正常使用？此导致了WaveGenerator.cpp中calcWave()计算步态时序的结果异常。
> 
> 运行环境是WIndows的Linux子系统（WSL2），环境就是原生Linux，唯一的区别就是显卡。能否留个联系方式，我把bug视频发给您看一下

您的师兄已经发过视频给我看了，我的环境里使用双系统，并且集显和独显(rtx 2060)都测试过都没出现这个问题，我们工程师还没有在WSL上开发过(曾经在B站看过相关视频说坑不少)，说实话我建议当下还是使用虚拟机或者装个双系统来使用。

closed 2023-03-08 by xiaoliangstd

sunflower-knight, 2024-08-18:
> ## 环境：
> * Windows10 Windows10的
> * WSL2最新版本内核
> * Ubuntu20.04最新版本
> * NVIDIA 3070 Laptop 8G NVIDIA 3070 笔记本电脑 8G
> * 显卡驱动最新版
> * ROS-noetic
> * Gazebo11 凉亭11
> 
> ## 问题
> 最开始在WSL2里使用gazebo仿真，此时步态正常，但发现帧率过低，故去Microsoft官网升级了WSLG，并同时更新了WSL2的内核与显卡驱动，升级前使用LLVM，如下图所示 [image] 升级后如下图，可以检测到独立显卡： [image] 但此时步态错乱，无法正常行走。 随后将同一份代码拷贝到VMware中同样环境下编译，步态正常，经查看VMware也是用LLVM，**遂怀疑此BUG与独立显卡有关**，下图为Vmware中的显卡信息 [image]

你好 这个问题的大概率是因为 wsl2 中显示驱动的 opengl 转义到 mesa ， mesa 转义到 window Direx12 ;  的过程中出现了问题。wsl2 中的mesa 版本太低，导致 gazebo 使用的 opengl4.3 或更新版本的指令 无法成功的转义到 Direx12 ； 
解决办法就是
1. 指定 wsl2 的gui 应用程序 使用 amd 显卡 如果你的 cpu 是 amd 核显或许可以试试。
2. 软件计算gazebo 中的3d 画面 但是很卡。 
3. 等待微软的wsl2 升级，在wslg 中提供更高版本 opengl 的默认支持。
4. 或许可以手动升级 wsl2 中的 mesa 驱动程序，不等了