unitreerobotics/unitree_guide #2: 勘误贴

fan-ziqi, 2023-03-06:
书中P82-式6.27$\omega_b^T$为向量，应为加粗小写字母$\boldsymbol{\omega_b^T}$，望再版时修正

xiaoliangstd, 2023-03-06:
好的，感谢反馈！已和出版社反馈。

1Liwenzhi, 2023-03-22:
书中79页的（6.13）$p_b$处应该表示的是机身坐标系{b}下足端相对于机身的位置坐标$p_{bfB}$

xiaoliangstd, 2023-03-22:
> 书中79页的（6.13）$ p_b $ 处应该表示的是机身坐标系{b}下足端相对于机身的位置坐标$p_{bfB}$

你好，非常感謝您的反馈！(6.13)式中的 $ p_b $ 确实应该改为 $ p_{bfB} $. 这里想表达的是由于机身旋转导致的足端速度. 我看了unitree_guide的代码，代码里是正确的，具体可以检查 unitreeRobot.cpp中的Vec34 QuadrupedRobot::getFeet2BVelocities(LowlevelState &state, FrameType frame)函数。再次感謝！

1Liwenzhi, 2023-03-26:
代码示例FeetEndcal.cpp文件中的第37行对应的是书中126页式（9.8），最后一项是K_w(w-w_d)，但是代码却写反了，是否有误？

fan-ziqi, 2023-04-05:
书中P109 `由式（6.10）可知，通过控制各个足端力......` ，这句话想引用的是否应该为式（6.40）？

fan-ziqi, 2023-04-06:
### 书中P91 第二段表述与前文不统一（强迫症，此条可忽略）
可见输出向量 $\boldsymbol{y}$ 的元素自上而下分别是世界坐标系{ $s$ }下四个足端~~到~~相对于机身的位置~~向量~~ $\boldsymbol{P}\_{sfB}$ ，世界坐标系{ $s$ }下四个足端~~到~~相对于机身的速度 $\boldsymbol{v}\_{sfB}$ ，......

fan-ziqi, 2023-04-06:
书中P95-式7.37 $y$ 为向量，应为加粗小写字母 $\boldsymbol{y}$ 。

fan-ziqi, 2023-04-08:
P133 代码块中有很多的多余空格

fan-ziqi, 2023-04-18:
P136 10.5 第三段_Kdw应为姿态阻尼

silencht, 2023-08-07:
P60，4.7节

> 本书只使用Eigen中**和**稠密矩阵（区别于稀疏矩阵）相关的部分

LiuXiaoyu-abc, 2024-04-13:
P77页：6.1.1“机身坐标系{b}实际上是一个静止的惯性系，只是机身坐标系{b}在任意时刻都与机器人的位姿重合”    
P78页：6.1.2"并且反复强调机身坐标系{b}是一个静止的坐标系，它并不随着机身一起运动，只是在当前时刻与机身的位姿重合"
以上两句话会不会冲突？？？
谢谢🙏

xiaoliangstd, 2024-04-16:
> P77页：6.1.1“机身坐标系{b}实际上是一个静止的惯性系，只是机身坐标系{b}在任意时刻都与机器人的位姿重合” P78页：6.1.2"并且反复强调机身坐标系{b}是一个静止的坐标系，它并不随着机身一起运动，只是在当前时刻与机身的位姿重合" 以上两句话会不会冲突？？？ 谢谢🙏

你好，我认为这并不冲突，之所以认为它是一个静止的惯性系，是因为牛顿第二定律在惯性系中才成立。同时书上也说了，此举是为了后面的牵连速度以及动力学运算。这是我的浅显理解。

Tipriest, 2025-05-26:
书中P84-式6.30最后一行 $w_s$ 为向量，应为加粗 $w_s$，望再版时修正

Tipriest, 2025-05-26:
书中P72-式5.38第二行$cos(\theta_{2})$求导有问题，求导的常数项$sin(\theta_{2})$上多了不应该有的导数号，望再版时修正

Tipriest, 2025-05-26:
第74页，摆动腿的控制小节，应该是目标速度，不是目标速度坐标

Tipriest, 2025-07-03:
> 书中78页式（6.5）应该是来源于本书参考文献【1】--现代机器人学 的式（3.73），是吗？若是的话，式（3.73）的描述是基于世界坐标系{s}，但本书式（6.5）中却是基于物体坐标系{b}，这一点，能解释一下吗？

......
你看一下现代机器人学里面的那个图，是一个刚体绕着s点旋转的，所以那个图里其实s是这本书里的机身坐标系的静止的中心b，而不是你说的世界坐标系