unitreerobotics/unitree_guide #6: 20.04编译成功，运行失败，大佬们能给出关于20.04跑代码的解决方案吗

2385861088, 2023-03-11:


xiaoliangstd, 2023-03-11:
你好，有运行失败的相关的视频或者截图？我的就是20.04跑的，并不一定需要18.04

2385861088, 2023-03-11:
> 你好，有运行失败的相关的视频或者截图？我的就是20.04跑的，并不一定需要18.04

gaobin@gaobin:~/diangou/unitree$ roslaunch unitree_guide gazeboSim.launch 
RLException: [gazeboSim.launch] is neither a launch file in package [unitree_guide] nor is [unitree_guide] a launch file name
The traceback for the exception was written to the log file

xiaoliangstd, 2023-03-11:
这个应该只是没有再ros工作空间下 source ./devel/setup.bash而已吧。再不行就把build文件删掉，重新编译，然后重新source ./develsetup.bash

2385861088, 2023-03-11:
好的，我试试，多谢大佬

xiaoliangstd, 2023-03-11:
客气了，有什么问题可以欢迎再提

2385861088, 2023-03-12:
> gaobin@gaobin:~/diangou/unitree$ source ./devel/setup.bash
gaobin@gaobin:~/diangou/unitree$ roslaunch unitree_guide gazeboSim.launch 
... logging to /home/gaobin/.ros/log/bb027d68-c0ed-11ed-8cbe-e599ff62cb0b/roslaunch-gaobin-3980.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: The following package was not found in <arg name="world_name" value="$(find unitree_gazebo)/worlds/$(arg wname).world"/>: unitree_gazebo
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/gaobin/diangou/unitree/src
ROS path [2]=/opt/ros/noetic/share

大佬，我这删除了build还是不行，src里就把guide那个功能包放在里面了。然后也source了，roslaunch也可以自动补齐

xiaoliangstd, 2023-03-12:
我觉得你还需要学习下 注意看终端提示： Resource not found: The following package was not found in : unitree_gazebo ， 就是你没有把unitree_ros包放在工作空间下。终端的信息已经提示为什么跑不通了。

2385861088, 2023-03-12:
> 

Open a terminal and switch the directory to the ros workspace containing unitree_guide, then run the following command to build the project:

catkin_make
当时看到这个以为只要一个包就行呢，多谢大佬，现在问题解决了

xiaoliangstd, 2023-03-12:
哈哈，客气了

dstx123, 2023-03-14:
请问你的可以正常运行吗，为什么我一运行之后狗站不稳啊，会直接摔倒

xiaoliangstd, 2023-03-15:
可以发一个视频，我这边分析下，我这边正常跑

closed 2023-03-15 by xiaoliangstd