unitreerobotics/unitree_guide #10: invoking "make -j8 -l8" failed

acoudgo, 2023-03-14:
在把united_guide，unitree_legged_msgs和unitree_ros_master一起放在catkin_ws/src中catkin_make后出现的报错，应该如何解决？rosdep check后无依赖缺失（all system dependencies have been satisfied）
在把united_move_base放在catkin_ws中单独catkin_make编译后也出现类似问题 但提示的是：running command:"make -j8 -l8"in"/catkin_ws/build"

xiaoliangstd, 2023-03-14:
你好，可以截一张您报错的终端图片我这边分析下

acoudgo, 2023-03-14:
[image]

xiaoliangstd, 2023-03-14:
图片我这边打开失败了，可以用preview看下显示正不正常

acoudgo, 2023-03-15:
[image]

acoudgo, 2023-03-15:
[image]

xiaoliangstd, 2023-03-15:
您这ubuntu 是用Mac os 跑的？

acoudgo, 2023-03-15:
我是有mac笔电 但这个ubuntu就是正常的主机安装的啊....好迷.....

xiaoliangstd, 2023-03-15:
终端信息为啥会有 “apple” "Mac ox x" 之类的信息呢

acoudgo, 2023-03-15:
我又去检查了一下cmake的安装版本 重新安装了一遍 确定了是cmake version 3.26.0 实在是想不通还有哪个环节会跟mac有关了

LJoson, 2023-03-15:
你不要用最大数量核数去catkin_make试试，由于cpu核心性能问题，ros编译过程中会出现这类问题

acoudgo, 2023-03-15:
我在win系统下重下了github上的文件 依然是invoking"make -j8 -l8"failed 但是报错和之前不一样 如下
[image]
然后 我又按照catkin_make -j1的方法 试了一次 是不是就是减少核数的方法 但还是报错 如下
[image]
请问还有什么方法可以处理这种情况

xiaoliangstd, 2023-03-15:
hello, look at the terminal, there is a hint "cannot find -llcm", that means you haven't installed the lcm. Please refer to this link https://github.com/lcm-proj/lcm to install. By the way, unitree_legged_real package should not be a part of dependencies，which is at the unitree_ros_to_real package.  这是在别的问题的回答，可以参考下，终端提示有为什么编译不通过

acoudgo, 2023-03-15:
[image]
在lcm安装过程进行到cmake ..这一步时 遇到了下面的问题 请问有遇到过类似的情况吗  暂时没找到
原链接：https://blog.csdn.net/weixin_45467056/article/details/123569027
[image]

xiaoliangstd, 2023-03-15:
这个请直接参考官方文档，其他的文章的情况我这边没遇到过，安装LCM的过程应该还是很简单的，不至与出错 ,  https://lcm-proj.github.io/lcm/content/build-instructions.html。

closed 2023-03-17 by xiaoliangstd