unitreerobotics/unitree_guide #8: 更改main.cpp出现的问题

Takagi0215, 2023-03-14:
想尝试一下不同的步态，在main.cpp将trot改为running_trot或pronk之后，再使用控制器控制机器人行走，出现崩溃

[image]

xiaoliangstd, 2023-03-14:
你好，对于unitree_guide的这个控制器，对于跟踪效果来说 目前最稳定的效果是trot步态。对于你选择的其它更为动态的步态，崩溃的原因有可能在与该控制器跟踪不上该步态下的期望，在某些构型下，算出的雅可比矩阵不太好，会出现nan的情况，导致崩溃。你可以尝试将WaveGenerator的步态周期参数改短，并增加支撑相的占空比。下图经供参考：
[image]

xiaoliangstd, 2023-03-14:
这是测试效果，该参数组合下 不至于崩溃。

https://user-images.githubusercontent.com/37007970/224885673-156e32f8-d687-480f-97dd-d3c70fb87438.mp4

Takagi0215, 2023-03-14:
谢谢大佬帮助！崩溃的问题已经解决！
还想请教一下，有没有比较好的bound步态的参数组合呢？我们试了一下，感觉不是很稳定

xiaoliangstd, 2023-03-14:
> 谢谢大佬帮助！崩溃的问题已经解决！ 还想请教一下，有没有比较好的bound步态的参数组合呢？我们试了一下，感觉不是很稳定

哈哈，这个我这边没怎么试过，因为涉及到实物的控制，所以调试最多的就是trot步态。您可以试着不光调步态，还可以调一下BalanceCtrl.cpp里面的 qp权重，或者试一下更高级的算法 例如MPC

closed 2023-03-14 by Takagi0215

Takagi0215, 2023-04-12:
您好，如果想要把MPC移植进来，有没有什么样例呢？

carryon68, 2024-08-29:
> 您好，如果想要把MPC移植进来，有没有什么样例呢？

您好，请问把MPC移植到这个上面，您解决了吗