unitreerobotics/unitree_guide #9: Ubuntu20.04正常运行，但是狗站立起来后会直接摔倒无法按键操控运动，请问是无法在20.04中运行吗

dstx123, 2023-03-14:


xiaoliangstd, 2023-03-14:
你好，我就是在20.04下跑的。这边还是建议你发个视频或者终端截图上来，这边仅凭文字无法判断

dstx123, 2023-03-14:
> 你好，我就是在20.04下跑的。这边还是建议你发个视频或者终端截图上来，这边仅凭文字无法判断

您好！这是我运行的结果，按了2之后站起来，然后按别的就直接摔倒了，请问这是为什么呀
https://user-images.githubusercontent.com/51892853/224923647-80164b47-61cf-4985-8d45-c907f35944e3.mp4

dstx123, 2023-03-14:
[image]
[image]
先按2之后能正常站起来，但是按了4之后就摔倒了

xiaoliangstd, 2023-03-14:
或者您还是录个视频吧，我这边看你的终端显示的信息，都是正常的

dstx123, 2023-03-14:
> > 你好，我就是在20.04下跑的。这边还是建议你发个视频或者终端截图上来，这边仅凭文字无法判断
> 
> 您好！这是我运行的结果，按了2之后站起来，然后按别的就直接摔倒了，请问这是为什么呀 https://user-images.githubusercontent.com/51892853/224923647-80164b47-61cf-4985-8d45-c907f35944e3.mp4

这个链接里面就是视频诶

xiaoliangstd, 2023-03-14:
我看终端还 提示 超时， 可以在前面加上 sudo ./devel/lib/unitree_guide/junior_ctrl , 这样可以把进程优先级提高，您的现在算一次花了5.2毫秒，就是约192hz，我们正常可以跑500hz

xiaoliangstd, 2023-03-14:
我看到了，您应该是用WSL来跑，我们没有在WSL上测试过，建议还是用虚拟机或者双系统跑吧。我猜测应该是步态程序里用的系统时间函数在WSL上用不了，有可能。

dstx123, 2023-03-14:
> 

不是用的wsl，这是用的向日葵连接的远程服务器。sudo ./devel/lib/unitree_guide/junior_ctrl 这个命令无法执行提示no such file or directory。去掉sudo可以正常运行

dstx123, 2023-03-14:
超时是在摔倒之后过了一会儿才发生的，我录个屏发给您看看吧

dstx123, 2023-03-14:
稍等一下，我压缩一下文件

xiaoliangstd, 2023-03-14:
想问下，您有试过在其他电脑跑嘛，会不会有这个问题

dstx123, 2023-03-14:
https://user-images.githubusercontent.com/51892853/224938137-89ce8f42-09e6-416c-bc84-f4de0c1c404a.mp4

视频j结果是这样的您看看，手头没有别的电脑诶

xiaoliangstd, 2023-03-14:
你好，我刚看了下你的视频，有一点疑惑，unitree_guide 的FSM里是有保护的即FSM.cpp 里面的checkSafty()函数，一般在机器人摔倒后，这个函数会返回true，从而让机器人从trotting 状态 转到 passive 状态即阻尼状态，此时机器人腿部会停下来。但从您的效果来看，这个函数明显没起作用，看下能否先debug这个问题。

xiaoliangstd, 2023-03-14:
至于它为什么会这样，我这边有朋友遇到过这个奇怪现象，我先和他咨询下，然后再答复你

dstx123, 2023-03-14:
麻烦您了！

dstx123, 2023-03-14:
会不会判断摔倒的传感器没有正常启动？我看xacro文件在melodic和neotic两个版本之间是有区别的。

xiaoliangstd, 2023-03-14:
> 会不会判断摔倒的传感器没有正常启动？我看xacro文件在melodic和neotic两个版本之间是有区别的。

应该不会，这个函数的判断是直接从IMU的姿态解算得到的旋转矩阵判断的。您那边方便在它翻转过来的时候把身体的旋转矩阵打印出来看下？

dstx123, 2023-03-14:
我该怎么打印呀，改哪个文件

xiaoliangstd, 2023-03-14:
bool FSM::checkSafty(){
    // The angle with z axis less than 60 degree
    std::cout<<"getRotMat: "<<std::endl;
    std::cout<<_ctrlComp->lowState->getRotMat()<<std::endl;
    std::cout<<std::endl;
    if(_ctrlComp->lowState->getRotMat()(2,2) < 0.5 ){
        return false;
    }else{
        return true;
    }
}

xiaoliangstd, 2023-03-14:
https://user-images.githubusercontent.com/37007970/224957535-bb0361bc-1256-4470-bc13-66b541214bc1.mp4

dstx123, 2023-03-14:
我这里这个数据没有变过诶

https://user-images.githubusercontent.com/51892853/224962021-4cf49903-8127-4e05-b256-d0f34103b429.mp4

dstx123, 2023-03-14:
一直都是单位矩阵

xiaoliangstd, 2023-03-14:
这个应该得一步步排查了，看下这里的函数会有变化嘛，里面的值
void IOROS::imuCallback(const sensor_msgs::Imu & msg)
{ 
    _lowState.imu.quaternion[0] = msg.orientation.w;
    _lowState.imu.quaternion[1] = msg.orientation.x;
    _lowState.imu.quaternion[2] = msg.orientation.y;
    _lowState.imu.quaternion[3] = msg.orientation.z;

    _lowState.imu.gyroscope[0] = msg.angular_velocity.x;
    _lowState.imu.gyroscope[1] = msg.angular_velocity.y;
    _lowState.imu.gyroscope[2] = msg.angular_velocity.z;
    
    _lowState.imu.accelerometer[0] = msg.linear_acceleration.x;
    _lowState.imu.accelerometer[1] = msg.linear_acceleration.y;
    _lowState.imu.accelerometer[2] = msg.linear_acceleration.z;
}

dstx123, 2023-03-14:
我尝试直接在函数里面std::cout没法输出，然后我看这个数据好像是作为/trunk_imu话题订阅的，我直接查看这个话题提示这样
[image]

dstx123, 2023-03-14:
我尝试按照网上说的rosparam set use_sim_time false执行后rostopic echo /trunk_imu会卡住，狗仍然会摔倒

xiaoliangstd, 2023-03-14:
好的，您的这个情况暂时我没有遇到，我这边先分析下解决方案

dstx123, 2023-03-14:
麻烦您了！

LJoson, 2023-03-15:
> 我看到了，您应该是用WSL来跑，我们没有在WSL上测试过，建议还是用虚拟机或者双系统跑吧。我猜测应该是步态程序里用的系统时间函数在WSL上用不了，有可能。

hello @xiaoliangstd , i use this repo in wsl2 but it don not have the wrong according to what u said

dstx123, 2023-03-15:
解决了，是ros环境的问题，导致没法正常加载imu。我卸载ros之后重新用鱼香ros一键安装就可以了。谢谢！

closed 2023-03-15 by dstx123

fan-ziqi, 2023-03-16:
# 已解决，附解决方法
#6 #1 
查看gazebo的imu数据，一切正常
[image]
但下图中话题未发布
[image]
遂怀疑gazebo与ros通信的插件有问题
在launch文件中的gazebo部分添加`<arg name="verbose" default="true"/>`，查看终端提示找不到插件
[image]
经过定位，缺少`ros-noetic-gazebo-plugins`、`ros-noetic-gazebo-ros-pkgs`这两个包
安装对应版本的包即可
```
sudo apt install ros-noetic-gazebo-ros-pkgs  ros-noetic-gazebo-msgs  ros-noetic-gazebo-plugins  ros-noetic-gazebo-ros-control
```

xiaoliangstd, 2023-03-17:
> # 已解决，附解决方法
>
> #6 #1 查看gazebo的imu数据，一切正常 [image] 但下图中话题未发布 [image] 遂怀疑gazebo与ros通信的插件有问题 在launch文件中的gazebo部分添加`<arg name="verbose" default="true"/>`，查看终端提示找不到插件 [image] [...]

强的大哥！

dstx123, 2023-03-17:
谢谢！

> # 已解决，附解决方法
> #6 #1 查看gazebo的imu数据，一切正常 [image] 但下图中话题未发布 [image] 遂怀疑gazebo与ros通信的插件有问题 在launch文件中的gazebo部分添加`<arg name="verbose" default="true"/>`，查看终端提示找不到插件 [image] [...]