unitreerobotics/unitree_guide #12: unitree_guide中找不到unitree_actuator_sdk的python版

LianzhaoZhang, 2023-03-16:
第7页中的内容说：“unitree_actuator_sdk都提供C，C++，Python以及ROS的代码实例"，但是没找到

xiaoliangstd, 2023-03-16:
你好，非常抱歉，unitree_actuator_sdk目前发现还有一点问题，近期(或一两天内)会更新在此仓库 (unitree_guide)， 该仓库仅仅定位配合该教材的使用。目前unitree_robotics 仓库下同名的unitree_actuator_sdk(https://github.com/unitreerobotics/unitree_actuator_sdk)是更简洁的电机sdk使用仓库 ,我们简化了其通信方式，但它目前只支持c++代码实例。

xiaoliangstd, 2023-03-16:
你好，非常抱歉，unitree_actuator_sdk目前发现还有一点问题，近期(或一两天内)会更新在此仓库 [(unitree_guide)](https://github.com/unitreerobotics/unitree_guide)， 该仓库仅仅定位配合该教材的使用。目前unitree_robotics 仓库下同名的unitree_actuator_sdk([https://github.com/unitreerobotics/unitree_actuator_sdk)](https://github.com/unitreerobotics/unitree_actuator_sdk) 是更简洁的电机sdk使用仓库,我们简化了其通信方式，但它目前只支持c++代码实例。

closed 2023-07-14 by xiaoliangstd