unitreerobotics/unitree_guide #14: 调用IOROS接口出现问题

Takagi0215, 2023-03-17:
想使用IOROS接口，编写简单程序对机器人关节进行控制。
为了简化，将IOROS.h中的sendCmd函数改为公共，如图。
[image]
在主程序中设置关节角度，尝试发布关节命令。
[image]
运行gazebo，进行仿真，错误：segmentation fault
[image]

xiaoliangstd, 2023-03-17:
这个我的建议是新建一个FSM，在 有限状态机里测试，而不是重写一个main.cpp

xiaoliangstd, 2023-03-17:
参考 State_SwingTest.cpp  / State_SwingTest.h

Takagi0215, 2023-03-19:
谢谢！但我还是想试着用guide里面的框架实现servo.cpp的功能，我觉得这样会更有助于我理解这个框架

xiaoliangstd, 2023-03-21:
还是不建议这样做，可以先基于现有的框架实现功能，然后再去改造现有的框架。如果实在需要这个，您可以借鉴一下先有的main.cpp文件。这个一般是程序里面有一些空指针(没有进行赋值，所以出错)，可以在代码里添加一些std::cout 语句，然后观察终端输出，看程序在那里输出中断了，然后一步步缩小范围确认在那里出错了。或者直接使用gdb进行调试。

closed 2023-03-22 by xiaoliangstd