unitreerobotics/unitree_guide #15: version of ubuntu and ros

nicolasukd, 2023-03-17:
Im trying to run this in ubuntu 22.04 and ros 2 humble. Do you think i can? have you tried this configuration?. Im asking cause you really should update this to newer versions of ros and ubuntu

xiaoliangstd, 2023-03-18:
Hello, very sorry that the unitree_guide does not support ros2, maybe it will be later. But if you are familiar with ros2, you can try to modify the project codes under unitree_guide first, such as interface/IOROS.h and interface/IOROS.cpp.

xiaoliangstd, 2023-03-18:
Hello, very sorry that the unitree_guide does not support ros2, maybe it will be later. But if you are familiar with ros2, you can try to modify the project codes of unitree_guide first, such as interface/IOROS.h and interface/IOROS.cpp.

closed 2023-03-21 by xiaoliangstd