unitreerobotics/unitree_guide #16: 关于FSM.cpp中的问题

Takagi0215, 2023-03-19:
学习到有限状态机的相关代码，其中检测机器人是否处于安全状态的checkSafty函数令我有些疑惑
`bool FSM::checkSafty(){
    // The angle with z axis less than 60 degree
    // 使用getRotMat函数可以获取当前姿态Rsb。
    if(_ctrlComp->lowState->getRotMat()(2,2) < 0.5 ){ 
        return false;
    }else{
        return true;
    }
}`
如果与z轴的角度少于60度，此时机器人是安全还是不安全呢？
为什么Rsb的第三行第三列小于0.5，具体是什么意思呢?

xiaoliangstd, 2023-03-19:
明天答复您，这个问题很容易理解的。

xiaoliangstd, 2023-03-21:
你好，首先_ctrlComp->lowState->getRotMat()函数得到的是一个旋转矩阵，该旋转矩阵是由IMU经过姿态解算得到的，是一个从body 坐标系 到 world 坐标系的旋转矩阵，即(from body to world, world 坐标系为参考坐标系 )。这里您可以简单理解为这个矩阵矩阵就是机器人躯干在世界坐标系下的姿态。” The angle with z axis less than 60 degree“的意思是机器人body坐标系上的z轴在世界坐标系下的投影小于0.5 就是 小于60度， 因为 cos(60度)等于0.5。通常情况下怎么会导致机器人body坐标系上的z轴在世界坐标系下的投影大于0.5呢，很大一部分可能就是机器人踏空，或者某时刻规划的力解算异常，导致机器人绕roll角 和 pitch 角旋转过大，达到一定程度此时就会大于0.5, 从机器人上看就是机器人快要倒了，次时这条函数就会返回 true，表示机器人是不安全的。那么FSM就会从当前状态机切换至阻尼(passive)状态来保护机器人。

closed 2023-03-21 by Takagi0215