unitreerobotics/unitree_guide #18: 学习IOROS过程中出现的问题

Alleneeous, 2023-03-21:
看书到三四五章，想调用sendRecv在ROS里发布一下关节角度，但是取观察不到gazobo中机器人的关节变化。以下是我写的测试代码：
[image]

请问是语法问题导致的嘛？还是我对sendRecv或是IOROS.cpp理解出现问题？或是其他原因？谢谢解答！

另外，我在IOROS.cpp中插入了若干的cout语句，全部能够打印（说明sendRecv函数被调用），却似乎没有任何关节消息发布
[image]

xiaoliangstd, 2023-03-21:
你好，我的建议是直接新建一个FSM，在一个专门的FSM状态机里测试您的功能，您这样的写法相当于重写了一个main.cpp了，这是我们不推荐的。您可以参考State_SwingTest.cpp 和 State_SwingTest.h 新建一个类似的用于测试您上述功能的状态机。

Alleneeous, 2023-03-21:
也就是说重新创建State_x.cpp和State_x.h，添加一个新的状态机模块，设置好启动该状态机的键位，再用jounir_ctrl打开自定义的状态机，这样理解正确吗？

xiaoliangstd, 2023-03-21:
是的，这样做的好处是 您就可以把主要的精力放于想要实现的功能上了，先实现您的功能剩下的事情就好办了。你上述问题没有响应需要参考下 
1.
    while(running){
        ctrlFrame.run();
    }
2.
void FSM::run(){
    _startTime = getSystemTime();
    _ctrlComp->sendRecv();
    _ctrlComp->runWaveGen();
    _ctrlComp->estimator->run();
    if(!checkSafty()){
        _ctrlComp->ioInter->setPassive();
    }

    if(_mode == FSMMode::NORMAL){
        _currentState->run();
        _nextStateName = _currentState->checkChange();
        if(_nextStateName != _currentState->_stateName){
            _mode = FSMMode::CHANGE;
            _nextState = getNextState(_nextStateName);
            std::cout << "Switched from " << _currentState->_stateNameString
                      << " to " << _nextState->_stateNameString << std::endl;
        }
    }
    else if(_mode == FSMMode::CHANGE){
        _currentState->exit();
        _currentState = _nextState;
        _currentState->enter();
        _mode = FSMMode::NORMAL;
        _currentState->run();
    }

    absoluteWait(_startTime, (long long)(_ctrlComp->dt * 1000000));
}

FSM Class 相当于一个状态机管理器，管理这FSM文件夹里的状态机。

closed 2023-03-22 by Alleneeous