unitreerobotics/unitree_guide #20: 关于控制框架移植

Alleneeous, 2023-03-23:
实验室使用GO-M8010-6 Motor搭建了一台四足机器人样机，需要把unitree_guide的控制框架移植到新的四足上（先进行仿真），请问需要进行哪些操作呢？

xiaoliangstd, 2023-03-23:
这个迁移工作需要您自己做吧，我们暂不提供这方面的支持。仅在课程范围之内。你可以充分学习课程内容、unitree_guide之后进行

closed 2023-03-24 by Alleneeous

fan-ziqi, 2023-04-23:
移植详见:
* https://github.com/fan-ziqi/cyberdog_sim
* https://github.com/fan-ziqi/unitree_guide#%E7%A7%BB%E6%A4%8D%E5%88%B0cyberdog
* https://github.com/fan-ziqi/unitree_ros/tree/main/robots/cyberdog_description

fan-ziqi, 2023-04-23:
> 实验室使用GO-M8010-6 Motor搭建了一台四足机器人样机，需要把unitree_guide的控制框架移植到新的四足上（先进行仿真），请问需要进行哪些操作呢？

移植详见:

* https://github.com/fan-ziqi/cyberdog_sim
* https://github.com/fan-ziqi/unitree_guide#%E7%A7%BB%E6%A4%8D%E5%88%B0cyberdog
* https://github.com/fan-ziqi/unitree_ros/tree/main/robots/cyberdog_description

Alleneeous, 2023-04-23:
十分感谢！