unitreerobotics/unitree_guide #21: 站立和前进狗会跳起的问题

KITKATXU, 2023-03-23:
https://user-images.githubusercontent.com/102399012/227129877-c1f9eccb-b1ab-4d0d-bd47-81117ac7e2aa.mp4

https://user-images.githubusercontent.com/102399012/227129933-40135ee4-4a2b-4abd-9b95-4ed92af06c7a.mp4

老师好，请问老师是否遇到过操作狗前进站立的时候狗经常向上跳一下导致站不稳的情况。站立的时候会跳起而不是平稳站起，前进的时候走一小段距离就会往后跳跃。我输出每次站立的start position感觉也没有问题，应该不是计算时没有获取到当前状态。

xiaoliangstd, 2023-03-23:
从你的第二个视频来看，你应该是用一台性能不是很好的电脑（或者当前电脑在并行很多高强度的计算任务）进行仿真，导致gazebo仿真的速率大大小于1.0，同时终端应该也会提示运算超时。所以还是换一台性能好的电脑试一下

KITKATXU, 2023-03-24:
老师您好，我的狗经常前进之后出现这个错误，且整个狗沉到地下，请问这也会跟我电脑性能不好有关吗？非常感谢~
[image]

xiaoliangstd, 2023-03-24:
我的建议是先换一台性能好的电脑测试一下。出现Nan的本质应该是笛卡尔空间下的足端力经过雅克比矩阵得到的关节空间力矩有问题，而unitree_guide中的雅可比矩阵跟机器人的各关节位置有关，当你电脑性能不好时，机器人各关节反馈的位置有延迟或者啥的，导致算得到的雅可比矩阵不太好，就应该会出现这个问题了。  从现象来看，应该是你的仿真页面看起来卡卡的，很慢

closed 2023-03-25 by xiaoliangstd

huaxi222, 2023-05-05:
大佬，我也是这个问题，运行tort步态的时候就Nan啦