unitreerobotics/unitree_guide #29: ros仿真导航功能

2385861088, 2023-04-03:
运行148页的，sudo su 之后运行ctrl程序，不能控制gazebo中狗子，进不去导航模式，之后运行导航程序的launch显示urdf模型缺失，只能显示电机关节

closed 2023-04-03 by 2385861088

dstx123, 2023-04-12:
您好，请问这个问题您是怎么解决的呀，我运行之后urdf模型就一直显示缺失，按5进入导航模式会有提示吗，我这儿也没显示诶

xiaoliangstd, 2023-04-12:
您可以上传一个终端显示图片，我们这边分析下

dstx123, 2023-04-12:
[image]
应该是TF坐标发布的不对，我没看到odom发布出来诶

dstx123, 2023-04-12:
[image]

xiaoliangstd, 2023-04-12:
这个是你没有安装 joint-state-publisher 这个包吧

dstx123, 2023-04-12:
只有这个导航的运行不了，但是unitree_guide里面的仿真都是能正常运行的，我看好像是odom不知道为什么没有发布出来

dstx123, 2023-04-12:
另外还有一个问题是，这个camera_face发不出来的图像为什么是倒着的呀

> 这个是你没有安装 joint-state-publisher 这个包吧

dstx123, 2023-04-12:
[image]

xiaoliangstd, 2023-04-12:
> [image]

这是可以了？

dstx123, 2023-04-12:
不是，这个是我自己写的另外的程序，这是另一个问题，就是发布出来的摄像头画面是倒着的。刚刚那个问题还是存在，应该是odom不知道为什么没有发布出来

> > [image]
> 
> 这是可以了？

xiaoliangstd, 2023-04-12:
请问你有按照书本上的安装语句把全部的包安装了？

dstx123, 2023-04-12:
> 请问你有按照书本上的安装语句把全部的包安装了？

我找到原因了，是因为cmakelists.txt中的MOVE_BASE没有设为ON，我改过来重新编译就可以了，谢谢！

dstx123, 2023-04-12:
[image]
但是还有个问题就是这个摄像头的画面为什么都是倒着的呀

xiaoliangstd, 2023-04-12:
> [image] 但是还有个问题就是这个摄像头的画面为什么都是倒着的呀

这个问题我有空排查下，您也可以尝试自己排查一下，估计是urdf 或者 某些参数设置原因？

dstx123, 2023-04-12:
> > [image] 但是还有个问题就是这个摄像头的画面为什么都是倒着的呀
> 
> 这个问题我有空排查下，您也可以尝试自己排查一下，估计是urdf 或者 某些参数设置原因？

我感觉可能是urdf的问题？因为我尝试在IsaacSim中使用这个urdf模型出来的摄像头画面也是反的。

xiaoliangstd, 2023-04-12:
> > > [image] 但是还有个问题就是这个摄像头的画面为什么都是倒着的呀
> > 
> > 
> > 这个问题我有空排查下，您也可以尝试自己排查一下，估计是urdf 或者 某些参数设置原因？
> 
> 我感觉可能是urdf的问题？因为我尝试在IsaacSim中使用这个urdf模型出来的摄像头画面也是反的。

这样的话，可以将urdf 里面连接的摄像机link的joint , 把 rpy 设置一下就行，调整到 正的画面

dstx123, 2023-04-12:
> > > > [image] 但是还有个问题就是这个摄像头的画面为什么都是倒着的呀
> > > 
> > > 
> > > 这个问题我有空排查下，您也可以尝试自己排查一下，估计是urdf 或者 某些参数设置原因？
> > 
> > 
> > 我感觉可能是urdf的问题？因为我尝试在IsaacSim中使用这个urdf模型出来的摄像头画面也是反的。
> 
> 这样的话，可以将urdf 里面连接的摄像机link的joint , 把 rpy 设置一下就行，调整到 正的画面

我在IsaacSim中这样尝试过，当摄像头画面是反的时候我建图出来是正的，但是当更改joint之后，摄像头画面正了建图却反了。我再试试看gazebo中会不会这样吧，也可能是isaacsim的问题。谢谢您！

XH-hlx, 2025-07-26:
> 不是，这个是我自己写的另外的程序，这是另一个问题，就是发布出来的摄像头画面是倒着的。刚刚那个问题还是存在，应该是odom不知道为什么没有发布出来
> 
> > > [image]
> > 
> > 
> > 这是可以了？

请问您可以分享一下建图的代码嘛，谢谢您，我想学习一下