unitreerobotics/unitree_guide #30: 安装时catkin_make报错

xuebuhuia123, 2023-04-06:
具体报错内容如下：
[ 64%] Linking CXX executable /home/yyh/catkin_ws/devel/lib/unitree_guide/junior_ctrl
//usr/local/lib/liblcm.so:无法识别文件: 文件被截断
collect2: error: ld returned 1 exit status
unitree_guide/CMakeFiles/junior_ctrl.dir/build.make:538: recipe for target '/home/yyh/catkin_ws/devel/lib/unitree_guide/junior_ctrl' failed
make[2]: *** [/home/yyh/catkin_ws/devel/lib/unitree_guide/junior_ctrl] Error 1
CMakeFiles/Makefile2:3458: recipe for target 'unitree_guide/CMakeFiles/junior_ctrl.dir/all' failed
make[1]: *** [unitree_guide/CMakeFiles/junior_ctrl.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
Consolidate compiler generated dependencies of target z1_controller
[ 67%] Built target z1_controller
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
是否有解决的方法

xiaoliangstd, 2023-04-06:
没有安装lcm?

xuebuhuia123, 2023-04-06:
非常感谢，已经解决问题了

xiaoliangstd, 2023-04-06:
好的，客气了

closed 2023-04-11 by xiaoliangstd