unitreerobotics/unitree_guide #31: 机器狗实体导航问题

sthkklog, 2023-04-07:
您好，请问一下书上152页机器狗实体导航是只适用与Go1狗吗？A1狗能用吗？能用的话需要做什么修改

xiaoliangstd, 2023-04-10:
您好，目前书上 152页是上的实体导航是只适用Go1机器人的。A1上如果想用的画需要将realsense相机的点云提取出来，替换对应的部分

sthkklog, 2023-04-10:
好的，谢谢

linjunbo123456, 2023-04-10:
请问在Trotting步态时，控制机器人运动的W A S D在程序那里可以找到

xiaoliangstd, 2023-04-10:
> 请问在Trotting步态时，控制机器人运动的W A S D在程序那里可以找到

你好，根据 W A S D控制机器人运动的本质是改变 期望的“参考轨迹”，这个可以在void State_Trotting::getUserCmd()函数里找到

linjunbo123456, 2023-04-10:
请问go1或着A1机器人的cmd_vel 速度话题在那里能找到呢，用rostopic list 和rqt_graph都没有办法能找到？

linjunbo123456, 2023-04-10:
请问go1/a1程序没有里程计odom吗？要是有，请问在那里可以找到呢？感谢

linjunbo123456, 2023-04-10:
请问在机器人按下2键能进入FixedStand，但为什么按5键没有响应，进不了导航模式move_base，这个有什么办法能解决呢

xiaoliangstd, 2023-04-11:
> 请问go1或着A1机器人的cmd_vel 速度话题在那里能找到呢，用rostopic list 和rqt_graph都没有办法能找到？

您这里说的是unitree_guide里的还是go1原厂程序呢

linjunbo123456, 2023-04-12:
unitree_guide 无法进入5模式

xiaoliangstd, 2023-04-12:
> unitree_guide 无法进入5模式

应该是你 set(MOVE_BASE ON)  这个没有设置为 ON 吧

linjunbo123456, 2023-04-12:
谢谢，宇树四足机器人导航功能包怎么调用？

xiaoliangstd, 2023-04-12:
> 谢谢，宇树四足机器人导航功能包怎么调用？
你说的是原厂程序还是unitree_guide的程序？ 需要区分下，unitree_guide是一个开源项目，和原厂配套的导航不一样

xiaoliangstd, 2023-04-12:
> unitree_guide的四足机器人导航功能包怎么调用

这个书本上就有说明的

closed 2023-07-17 by xiaoliangstd