unitreerobotics/unitree_guide #36: 仿真验证算法3.5.3。p33

Lixiaolong310, 2023-04-22:
运行rosrun unitree_guide junior_ctrl
[ERROR] Function setProcessScheduler failed.
The control interface for ROS Gazebo simulation
robot_name: a1
Switched from passive to fixed stand
没有反应，站不起来

xiaoliangstd, 2023-04-23:
你好，站不起来的原因有很多种，需要您提供更多的一些信息来供我们分析一下，可以上传些视频或者图片

mylinfh, 2023-07-11:
[image]

mylinfh, 2023-07-11:
请问这是什么原因呢https://github.com/unitreerobotics/unitree_guide/issues/36#issuecomment-1518916824

mylinfh, 2023-07-11:
请问catkin_make时出现这种错误该怎么办呢
[image]

mylinfh, 2023-07-11:
> 你好，站不起来的原因有很多种，需要您提供更多的一些信息来供我们分析一下，可以上传些视频或者图片
请问catkin_make时出现这种错误该怎么办呢
[image]

xiaoliangstd, 2023-07-17:
> > 你好，站不起来的原因有很多种，需要您提供更多的一些信息来供我们分析一下，可以上传些视频或者图片
> > 请问catkin_make时出现这种错误该怎么办呢
> > [image]

这里应该是你没有 catkin_make 吧

yccckid, 2024-05-26:
您好，为什么站起来之后，按4，马上就倒了呢？

jiaweinie, 2024-09-12:
> 您好，为什么站起来之后，按4，马上就倒了呢？

请问你现在解决这个问题了么