unitreerobotics/unitree_guide #37: 在ROS中仿真导航

xuebuhuia123, 2023-04-24:
按下键盘上的5进入move_base模式后，只要用新的终端运行启动RViz的代码，运行Gazebo的终端就会报错，然后运行RViz的终端会出很多警告，有办法解决吗
[image]

xiaoliangstd, 2023-04-27:
你好，关于这个可以再提供多一些截图我看看，或者也可以上翻终端截图，一般是没有安装某些库导致的

xuebuhuia123, 2023-04-27:
只运行了三个终端，在打开RViz之后，第一个驱动Gazebo的终端就会直接提示进程被杀死像我截图里右边红色的字体那样，然后第二个在root下运行的指令还是正常的，RViz会有图片里左边那样的警告，但是RViz也可以正常运行，只不过RViz里只能看到机器狗，并不能看到环境信息，现在RViz里机器狗还会出现身体和四肢脱离的情况。我尝试把Gazebo从9.0.0版本升级到9.19.0，但是也没有解决问题，而且书上提到的依赖我也都安装好了，catkin_make是没有报错的，Move_base也设置了ON。
[image]

Gazebo报错代码如下：
Killed
[gazebo-2] process has died [pid 4139, exit code 137, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode /home/yyh/catkin_ws/src/unitree_move_base/worlds/smallRoom.world __name:=gazebo __log:=/home/yyh/.ros/log/dd73e3fc-e4ae-11ed-a52e-000c294656d2/gazebo-2.log].
log file: /home/yyh/.ros/log/dd73e3fc-e4ae-11ed-a52e-000c294656d2/gazebo-2*.log
RViz警告代码如下：
[ WARN] [1682567638.393247527, 849.356000000]: The /rightLaserScan observation buffer has not been updated for 849.36 seconds, and it should be updated every 3.30 seconds.
 WARN] [1682567617.368556134, 846.801000000]: local_costmap: Parameter "plugins" not provided, loading pre-Hydro parameters
而且根据书中所说，在Root下执行控制代码，是不会出现警告的，但是我这里还是出现了
root@yyh-virtual-machine:/home/yyh# rosrun unitree_guide junior_ctrl
The control interface for ROS Gazebo simulation
robot_name: go1
Switched from passive to fixed stand
Switched from fixed stand to move_base
[WARNING] The waitTime=2000 of function absoluteWait is not enough!
The program has already cost 3575us.
[WARNING] The waitTime=2000 of function absoluteWait is not enough!
The program has already cost 4082us.

xiaoliangstd, 2023-04-27:
一般来说，
Gazebo报错代码如下：
Killed
[gazebo-2] process has died [pid 4139, exit code 137, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode /home/yyh/catkin_ws/src/unitree_move_base/worlds/smallRoom.world __name:=gazebo __log:=/home/yyh/.ros/log/dd73e3fc-e4ae-11ed-a52e-000c294656d2/gazebo-2.log].
log file: /home/yyh/.ros/log/dd73e3fc-e4ae-11ed-a52e-000c294656d2/gazebo-2*.log，
这段代码 上面肯定还是会有一些其他的红色文字，或者其他的信息，这个信息才能说明是什么库没找到才会出现 下面的 “process has died”

xuebuhuia123, 2023-04-27:
下面是终端运行的全部信息，您看一下，
yyh@yyh-virtual-machine:~$ cd catkin_ws
yyh@yyh-virtual-machine:~/catkin_ws$ roslaunch unitree_move_base gazebo_move_base.launch
... logging to /home/yyh/.ros/log/16f04dda-e4cf-11ed-871d-000c294656d2/roslaunch-yyh-virtual-machine-2549.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://yyh-virtual-machine:33015/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /go1_gazebo/FL_calf_controller/joint: FL_calf_joint
 * /go1_gazebo/FL_calf_controller/pid/d: 8.0
 * /go1_gazebo/FL_calf_controller/pid/i: 0.0
 * /go1_gazebo/FL_calf_controller/pid/p: 300.0
 * /go1_gazebo/FL_calf_controller/type: unitree_legged_co...
 * /go1_gazebo/FL_hip_controller/joint: FL_hip_joint
 * /go1_gazebo/FL_hip_controller/pid/d: 5.0
 * /go1_gazebo/FL_hip_controller/pid/i: 0.0
 * /go1_gazebo/FL_hip_controller/pid/p: 100.0
 * /go1_gazebo/FL_hip_controller/type: unitree_legged_co...
 * /go1_gazebo/FL_thigh_controller/joint: FL_thigh_joint
 * /go1_gazebo/FL_thigh_controller/pid/d: 8.0
 * /go1_gazebo/FL_thigh_controller/pid/i: 0.0
 * /go1_gazebo/FL_thigh_controller/pid/p: 300.0
 * /go1_gazebo/FL_thigh_controller/type: unitree_legged_co...
 * /go1_gazebo/FR_calf_controller/joint: FR_calf_joint
 * /go1_gazebo/FR_calf_controller/pid/d: 8.0
 * /go1_gazebo/FR_calf_controller/pid/i: 0.0
 * /go1_gazebo/FR_calf_controller/pid/p: 300.0
 * /go1_gazebo/FR_calf_controller/type: unitree_legged_co...
 * /go1_gazebo/FR_hip_controller/joint: FR_hip_joint
 * /go1_gazebo/FR_hip_controller/pid/d: 5.0
 * /go1_gazebo/FR_hip_controller/pid/i: 0.0
 * /go1_gazebo/FR_hip_controller/pid/p: 100.0
 * /go1_gazebo/FR_hip_controller/type: unitree_legged_co...
 * /go1_gazebo/FR_thigh_controller/joint: FR_thigh_joint
 * /go1_gazebo/FR_thigh_controller/pid/d: 8.0
 * /go1_gazebo/FR_thigh_controller/pid/i: 0.0
 * /go1_gazebo/FR_thigh_controller/pid/p: 300.0
 * /go1_gazebo/FR_thigh_controller/type: unitree_legged_co...
 * /go1_gazebo/RL_calf_controller/joint: RL_calf_joint
 * /go1_gazebo/RL_calf_controller/pid/d: 8.0
 * /go1_gazebo/RL_calf_controller/pid/i: 0.0
 * /go1_gazebo/RL_calf_controller/pid/p: 300.0
 * /go1_gazebo/RL_calf_controller/type: unitree_legged_co...
 * /go1_gazebo/RL_hip_controller/joint: RL_hip_joint
 * /go1_gazebo/RL_hip_controller/pid/d: 5.0
 * /go1_gazebo/RL_hip_controller/pid/i: 0.0
 * /go1_gazebo/RL_hip_controller/pid/p: 100.0
 * /go1_gazebo/RL_hip_controller/type: unitree_legged_co...
 * /go1_gazebo/RL_thigh_controller/joint: RL_thigh_joint
 * /go1_gazebo/RL_thigh_controller/pid/d: 8.0
 * /go1_gazebo/RL_thigh_controller/pid/i: 0.0
 * /go1_gazebo/RL_thigh_controller/pid/p: 300.0
 * /go1_gazebo/RL_thigh_controller/type: unitree_legged_co...
 * /go1_gazebo/RR_calf_controller/joint: RR_calf_joint
 * /go1_gazebo/RR_calf_controller/pid/d: 8.0
 * /go1_gazebo/RR_calf_controller/pid/i: 0.0
 * /go1_gazebo/RR_calf_controller/pid/p: 300.0
 * /go1_gazebo/RR_calf_controller/type: unitree_legged_co...
 * /go1_gazebo/RR_hip_controller/joint: RR_hip_joint
 * /go1_gazebo/RR_hip_controller/pid/d: 5.0
 * /go1_gazebo/RR_hip_controller/pid/i: 0.0
 * /go1_gazebo/RR_hip_controller/pid/p: 100.0
[... 77 lines cut ...]
[INFO] [1682581368.253011, 828.073000]: Loading controller: FL_hip_controller
[INFO] [1682581368.374106, 828.090000]: Loading controller: FL_thigh_controller
[INFO] [1682581368.390499, 828.099000]: Loading controller: FL_calf_controller
[INFO] [1682581368.409216, 828.109000]: Loading controller: FR_hip_controller
[INFO] [1682581368.427458, 828.121000]: Loading controller: FR_thigh_controller
[INFO] [1682581368.437583, 828.122000]: Loading controller: FR_calf_controller
[INFO] [1682581368.462715, 828.133000]: Loading controller: RL_hip_controller
[INFO] [1682581368.485537, 828.135000]: Loading controller: RL_thigh_controller
[INFO] [1682581368.508871, 828.148000]: Loading controller: RL_calf_controller
[INFO] [1682581368.524701, 828.149000]: Loading controller: RR_hip_controller
[INFO] [1682581368.548373, 828.155000]: Loading controller: RR_thigh_controller
[INFO] [1682581368.568902, 828.155000]: Loading controller: RR_calf_controller
[INFO] [1682581368.597016, 828.173000]: Controller Spawner: Loaded controllers: joint_state_controller, FL_hip_controller, FL_thigh_controller, FL_calf_controller, FR_hip_controller, FR_thigh_controller, FR_calf_controller, RL_hip_controller, RL_thigh_controller, RL_calf_controller, RR_hip_controller, RR_thigh_controller, RR_calf_controller
[INFO] [1682581368.603800, 828.173000]: Started controllers: joint_state_controller, FL_hip_controller, FL_thigh_controller, FL_calf_controller, FR_hip_controller, FR_thigh_controller, FR_calf_controller, RL_hip_controller, RL_thigh_controller, RL_calf_controller, RR_hip_controller, RR_thigh_controller, RR_calf_controller
[ INFO] [1682581378.018908123, 834.778000000]: Load RR_foot_contact Draw Force plugin.
[ INFO] [1682581378.029723677, 834.785000000]: Load RL_foot_contact Draw Force plugin.
[ INFO] [1682581378.035640541, 834.790000000]: Load FR_foot_contact Draw Force plugin.
[ INFO] [1682581378.049750227, 834.800000000]: Load FL_foot_contact Draw Force plugin.
[ INFO] [1682581388.433365641, 840.465000000]: Load RR_foot_contact Draw Force plugin.
[ INFO] [1682581388.469525719, 840.484000000]: Load RL_foot_contact Draw Force plugin.
[... cut ...] 15, 840.492000000]: Load FL_foot_contact Draw Force plugin.
[ INFO] [1682581388.502131578, 840.498000000]: Load FR_foot_contact Draw Force plugin.
Killed
[gazebo-2] process has died [pid 2581, exit code 137, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -u -e ode /home/yyh/catkin_ws/src/unitree_move_base/worlds/smallRoom.world __name:=gazebo __log:=/home/yyh/.ros/log/16f04dda-e4cf-11ed-871d-000c294656d2/gazebo-2.log].
log file: /home/yyh/.ros/log/16f04dda-e4cf-11ed-871d-000c294656d2/gazebo-2*.log

xuebuhuia123, 2023-05-06:
我在终端上看到除了杀死进程的报错以外，上面没有别的红色的字体报错

xuebuhuia123, 2023-05-06:
您好，现状软件崩溃问题已经解决了，但是在RViz上给定一个目标点后，会出现如下报错怎么办：
[ERROR] [1683369335.684314316, 873.872000000]: Ignoring transform for child_frame_id "base" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (0.000000 0.000000 0.000000 1.000000)

RViz上的图像是这样的
[image]

[image]

xiaoliangstd, 2023-05-06:
您最后这张图的机器狗呢？

xiaoliangstd, 2023-05-06:
方便的化，可以录屏一个，这边想看看全过程

xiaoliangstd, 2023-05-06:
我猜测 gazebo 仿真里的机器狗是不是只剩下躯干了，四腿都不见了

xuebuhuia123, 2023-05-06:
对，稍等一下，我录视频给您看一下

xuebuhuia123, 2023-05-06:
https://user-images.githubusercontent.com/130046140/236620665-746f1172-b16c-474b-89cd-ad7246c4f3ea.mp4

xiaoliangstd, 2023-05-06:
这个怎么感觉是您电脑或者虚拟机配置的性能太低的原因呢？

xuebuhuia123, 2023-05-06:
虚拟机配置是这样的，您看一下够用吗
[image]

xiaoliangstd, 2023-05-06:
您可以在您同事或者同学的 装双系统的电脑上试一下这个工程先，，看您的视频 gazebo里的机器狗动的太慢了，明显有点吃力。 这里会导致状态估计器乱飘了，所以rviz的机器狗会乱动，状态故机器飘了，导致力优化出问题，会算出一些不符合常理的力，这些力转换后直接送入关节控制器，导致了崩溃。 有可能是因为这样。最好再性能号一点的电脑 双系统上测试

xuebuhuia123, 2023-05-06:
好的，谢谢您，我后续再试试

xiaoliangstd, 2023-05-06:
我对电脑这方面也不太熟悉，但是正常gazebo里显示的帧率不太会这么低的

xuebuhuia123, 2023-05-06:
我运行的时候帧率是有50~60帧左右的

xuebuhuia123, 2023-05-06:
那我换个设备先试试把

xiaoliangstd, 2023-05-06:
哈哈没 ，看你的gazebo 的 real time factor 跑起来 直接到了 0.0几的这个数字，，这个太低了，，正常我们都是 0.9 这样，就是比较接近现实时间。

xuebuhuia123, 2023-05-06:
OK，我这几天换个设备试试，谢谢您的答疑

xiaoliangstd, 2023-05-06:
好的，客气了，直接叫您的同事或者同学测试下就行。

xiaoliangstd, 2023-05-06:
我新建了个群，有啥问题可以在群里沟通哈

[image]

dbdxnuliba, 2023-05-24:
@xiaoliangstd ,您好，答疑群已经过期了，方便发个新的么

LianzhaoZhang, 2023-07-28:
@xiaoliangstd 您好，答疑群已经过期了，方便发个新的么

vchint6, 2023-09-28:
> 按下键盘上的5进入move_base模式后，只要用新的终端运行启动RViz的代码，运行Gazebo的终端就会报错，然后运行RViz的终端会出很多警告，有办法解决吗 [image]

Hello i am working on unitree go1. I am unable to change Fixed frame in Rviz to Odom as it is not being published. How can I publish odom topic?

dmw0212, 2025-05-07:
@xiaoliangstd  可以再发一下群吗？