unitreerobotics/unitree_guide #39: pyplot的问题

XiaoJichun, 2023-04-26:
请问将CMakeList中的DEBUG设置为ON后，只能使用python2吗？我的版本是3.8.10，怎么可以直接使用python3进行使用呢？有更改方法吗？
[image]

[image]
在我catkin_make时，出现了以下问题：
[image]

XiaoJichun, 2023-05-08:
为啥子不回答我

xiaoliangstd, 2023-05-10:
您好，不好意思，没注意到，因为以前也有人问过这个问题，这个问题我近段时间会集中处理下，您也可以加一下这个群，群里有其他大佬解决过这个问题：

[image]

closed 2023-07-18 by xiaoliangstd