unitreerobotics/unitree_guide #42: 控制器接受速度命令

nanbwrn, 2023-05-06:
请问，base_local_planner模块发布的速度命令 /cmd_vel 在哪里能找到哪 
[image]

xiaoliangstd, 2023-05-06:
你好，这个是根据相应的base_local_planner包来发布的，具体可以看一下 roscd base_local_planner, 具体发布的话题，unitree_guide里仅仅是订阅，这个包发布什么 这边订阅后就执行了

nanbwrn, 2023-05-06:
您说的是这个吗，我在这没找到发布的具体话题，
[image]

nanbwrn, 2023-05-06:
我想这模仿着move_base 里的操作把State_Trotting中的摇杆命令覆盖掉，用/cmd_vel发布的命令来运动，现在找到了订阅操作了，我打算再模仿着，进行发布者实现

nanbwrn, 2023-05-06:
我把发布的速度加在了main.cpp中，并且在State_Tortting类中，覆盖了getUserCmd,最终运行的时候，发现gazebo中go1不能按照发布的速度指令运动。
[image]
[image]
[image]
[image]

https://user-images.githubusercontent.com/132453617/236619185-559b588a-a7d7-4cd4-8e64-5d704009d01e.mp4

xiaoliangstd, 2023-05-06:
你好，那是因为在 State_Trotting里没有接收这个话题后对应的行为，你可以在 calcCmd() 函数里把你接收到的数据，用来修改 _velBody的值，这里修改相当于修改了“参考轨迹”，控制器感受到参考轨迹后，才可以优化出力 去驱动躯干 尽量地跟上参考轨迹。而 原来State_Trotting是没有代码来修改这个参考轨迹的

xiaoliangstd, 2023-05-06:
例如 
在calcCmd()函数里 头部位置加上
_velBody(0) = _vx;
_velBody(1) = _vy;

nanbwrn, 2023-05-06:
您好，我在在calcCmd()函数里 头部位置加上_velBody(0) 和_velBody(1)  之后，控制器好像还是没有感受到参考轨迹

[image]

xiaoliangstd, 2023-05-06:
正常来说不会的，你打印出来看看

nanbwrn, 2023-05-06:
是在回调函数里写一个打印语句吗

nanbwrn, 2023-05-06:
谢谢老师，我解决了

xiaoliangstd, 2023-05-06:
[image]

xiaoliangstd, 2023-05-06:
我新建了个群，有啥问题可以在群里提哈，方便沟通

closed 2023-06-09 by xiaoliangstd