unitreerobotics/unitree_guide #52: 关于build的问题

1874ruanshaodong, 2023-07-09:
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  unitree_guide/unitree_move_base/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rsd/guide_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rsd/guide_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

在catkin_make后出现以上错误，这是什么问题呢？

1874ruanshaodong, 2023-07-10:
解决了，原来没有安装包，
sudo apt-get install ros-melodic-move-base-msgs

xiaoliangstd, 2023-07-11:
你好，是的，这就是由于move-base的东西没有安装

closed 2023-07-13 by xiaoliangstd