unitreerobotics/unitree_guide #58: 请问能再发一下交流群链接吗 图片过期了

liuz200103, 2023-08-11:
你好
[image]

_Originally posted by @xiaoliangstd in https://github.com/unitreerobotics/unitree_guide/issues/56#issuecomment-1661937881_

xiaoliangstd, 2023-08-11:
[image]

an1ya1, 2023-08-28:
能再发一下最新的吗？

xiaoliangstd, 2023-09-13:
[image]

closed 2023-09-27 by xiaoliangstd

Jiang-sd, 2023-10-18:
交流群链接过期了，可以更新一下吗

Jason3318, 2023-10-29:
Hello, the QR code for the communication group has expired. Could you please update the QR code again?

xiaoliangstd, 2023-10-30:
> 交流群链接过期了，可以更新一下吗
[image]

xiaoliangstd, 2023-10-30:
> Hello, the QR code for the communication group has expired. Could you please update the QR code again?

[image]

an1ya1, 2024-11-05:
> Can you send the QR code again? I want to join.

不介意的话，你可以发下vx号到我邮箱。这个群已经满了，我加你然后邀请你进群。

HuanLiu-Christian, 2024-11-05:
你好，
       我微信号是lh674539824，非常感谢！

an1ya1, 2024-11-05:
> 你好，
>        我微信号是lh674539824，非常感谢！
> 
> 
> > Can you send the QR code again? I want to join.
> >
> > 不介意的话，你可以发下vx号到我邮箱。这个群已经满了，我加你然后邀请你进群。

你好，我微信已经加你了。我是生如夏花