unitreerobotics/unitree_guide #65: catkin_make 编译时错误

LIHUA919, 2023-12-05:
[image]
[image]

zhj3180, 2024-01-08:
兄弟你解决问题了吗，我的解决方案是
sudo apt-get install ros-noetic-navigation

希望能帮到你

yannyann11, 2024-02-07:
大家好啊，我这个编译catkin_make时 unitree_guide/junior_ctrl这边一直报错呢

[image]

2422295290, 2024-03-14:
你好我也是这样，请问解决了么

xiaoliangstd, 2024-03-18:
> 大家好啊，我这个编译catkin_make时 unitree_guide/junior_ctrl这边一直报错呢
> 
> <img alt="380c4ee07b1f9e39461e81f36617be0" width="362" src="https://private-user-images.githubusercontent.com/74200027/302849639-669cdab3-e214-4d98-a57b-817dfa1f5b96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MzA0ODAsIm5iZiI6MTcxMDczMDE4MCwicGF0aCI6Ii83NDIwMDAyNy8zMDI4NDk2MzktNjY5Y2RhYjMtZTIxNC00ZDk4LWE1N2ItODE3ZGZhMWY1Yjk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyNDk0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFkNjNiOGY4ZTUyYThjZDVmZWVhNmY5MThmZTc3ZjBlY2Q1ZmZhNDZiNTMyYzE0Y2Q2MjI0NGQ4OTdlOWY1OGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QMWNr-tIV9UwXOYtLDWHPnwPRXJ3o4ud1QL-kjuEmec">

编译问题可以加入此群：
[image]

xiaoliangstd, 2024-03-18:
> [ 17%] Built target actionlib_msgs_generate_messages_nodejs​/usr/bin/ld: 当搜索用于 /home/ros/catkin_ws/src/unitree_guide/unitree_actuator_sdk/unitree_motor_ctrl/lib/libUnitree_motor_SDK_Linux.so 时跳过不兼容的 -lUnitree_motor_SDK_Linux​/usr/bin/ld: 找不到 -lUnitree_motor_SDK_Linux​collect2: error: ld returned 1 exit status
> 
> 请问什么时候在这里也能够加上arm64版本的.so文件呢？现在的情况来看在arm64架构的机器上无法正常的编译运行 ![image](https://private-user-images.githubusercontent.com/108170542/313411209-63b5f9b1-2ab8-4cc1-8ff3-c412cd265d7f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MzA0ODAsIm5iZiI6MTcxMDczMDE4MCwicGF0aCI6Ii8xMDgxNzA1NDIvMzEzNDExMjA5LTYzYjVmOWIxLTJhYjgtNGNjMS04ZmYzLWM0MTJjZDI2NWQ3Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxOFQwMjQ5NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01Y2QzYjQ4NmQwYjdjZjFiNTcxYmRlNWYzMmY1ZmM0MzEzY2U2NjE0ZDhkNWE2NmQ0ZmZkNTI0ZDg0ZTFjZDQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Ff3egp535SYQ-63ROP_NQ1VtVZeyuhmpCejdTk-acfE)

unitree_guide 里的 unitree_actuator_sdk 不打算更新了，如果有编译问题，可以直接将此库删除，此库是独立的

Jocab0602, 2024-03-28:
群聊图片过期了

xiaoliangstd, 2024-04-23:
> 群聊图片过期了

您好，有问题的话，可以发邮件给我们，我们也会即时回复的

closed 2024-04-23 by xiaoliangstd

shaashou, 2025-06-18:
同样的报错也不是在arm架构下面呀