unitreerobotics/unitree_guide #72: gazebo掉帧

KK-Zhou, 2024-04-18:
正常运行空的gazebo大约有60帧，运行unitree_guide的gazeboSim.launch就不到3帧了
系统：ubuntu20.04.06
显卡：NVIDIA RTX 4090

xiaoliangstd, 2024-04-18:
请问您的运行截图有嘛，可以发一个我们判断一下

KK-Zhou, 2024-04-19:
[image]
FPS很低

KK-Zhou, 2024-04-19:
> 请问您的运行截图有嘛，可以发一个我们判断一下

[image]
同时没有对gzclient与gzserver的显存占用

xiaoliangstd, 2024-04-19:
那你应该没有开启显卡性能模式，正常4090显卡运行gazebo 时，环境中会有太阳光渲染。可参考 https://blog.csdn.net/weixin_44001261/article/details/109141169  中的 安装方法一 第5步 运行   nvidia-settings ，还有第6步 显卡模式改成性能模式。

xiaoliangstd, 2024-04-19:
这是4080 运行画面，此时开启显卡性能模式效果
[image]

xiaoliangstd, 2024-04-22:
> > 这是4080 运行画面，此是开启显卡性能模式效果 ![Screenshot from 2024-04-19 10-05-02](https://private-user-images.githubusercontent.com/37007970/323809294-eabdb5d1-0179-4041-98b8-272dcb924322.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1OTU1MjYsIm5iZiI6MTcxMzU5NTIyNiwicGF0aCI6Ii8zNzAwNzk3MC8zMjM4MDkyOTQtZWFiZGI1ZDEtMDE3OS00MDQxLTk4YjgtMjcyZGNiOTI0MzIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIwVDA2NDAyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4M2JhZDlhZGIxMzQ3ZmRkYzM2OGRmYzg0ZjRhOTRmODJjNmEzOTEwMjkwY2ZjMDY4NGRhNjI3NzI4NzFjMjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SsVb05zkMh1kMrGxyiVdcnHKi-L2j9e8k8CP8FkYI2A)
> 
> 感谢您的解答，该问题是我没有将gazebo运行在独显上导致无法利用GPU加速，目前已解决

客气了

closed 2024-04-22 by xiaoliangstd