unitreerobotics/unitree_guide #73: 能再发下交流群群号吗？亲~~

abloomyt, 2024-04-22:
可能有些问题想问

xiaoliangstd, 2024-04-22:
你好，您可以直接在issues里提问，我们也会即时回答的

abloomyt, 2024-04-23:
谢谢你的回复！那我就不客气了！
State_Trotting.cpp中触地脚的力控制程序关于角度那部分：
_yawCmd = _yawCmd + _dYawCmd * _ctrlComp->dt;
_Rd = rotz(_yawCmd);
_dWbd  = _kpw*rotMatToExp(_Rd*_G2B_RotMat) + _Kdw * (_wCmdGlobal - _lowState->getGyroGlobal());
这里的_Rd的意思是不是目标姿态与0时刻姿态之间的差？但书中的意思是目标姿态与当前（上一时刻）姿态之的差，这里我有点不太明白

xiaoliangstd, 2024-04-23:
> 谢谢你的回复！那我就不客气了！ State_Trotting.cpp中触地脚的力控制程序关于角度那部分： _yawCmd = _yawCmd + _dYawCmd * _ctrlComp->dt; _Rd = rotz(_yawCmd); _dWbd = _kpw_rotMatToExp(_Rd__G2B_RotMat) + _Kdw * (_wCmdGlobal - _lowState->getGyroGlobal()); 这里的_Rd的意思是不是目标姿态与0时刻姿态之间的差？但书中的意思是目标姿态与当前（上一时刻）姿态之的差，这里我有点不太明白

从仿真里看应该是世界坐标系下的期望姿态。

xiaoliangstd, 2024-04-23:
你可以尝试按这样的方式改一下_Rd, 然后重新编译运行，就会发现它始终会保持和世界坐标系水平。
[image]

abloomyt, 2024-04-23:
嗯嗯，但是书的意思是期望姿态相对于当前姿态的差，所以我有点误解
[image]

abloomyt, 2024-04-23:
但是这两个姿态差不是同一个差吧，
1.代码中是相对于世界坐标系下的姿态差R1
2.书中是相对于当前姿态的姿态差R2，当前姿态不一定与世界坐标系重合吧

xiaoliangstd, 2024-04-23:
> 但是这两个姿态差不是同一个差吧， 1.代码中是相对于世界坐标系下的姿态差R1 2.书中是相对于当前姿态的姿态差R2，当前姿态不一定与世界坐标系重合吧

按你的理解，应该是 Rsb * R = Rd ，中的 R , 才是 姿态差嘛

abloomyt, 2024-04-23:
我明白都是姿态差，但是我觉得书上的姿态差和代码上的姿态差不是同一个姿态差

closed 2024-09-07 by abloomyt