unitreerobotics/unitree_guide #75: 找不到11.5.2节中unitree_pointCloud_publisher的程序包，有网址吗？

Panting007, 2024-05-13:


Panting007, 2024-06-03:
> 你好，请问你有邮箱嘛，我把这个包发你

你好，我的邮箱是pantene0807@163.com

Panting007, 2024-06-17:
> 你好，请问你有邮箱嘛，我把这个包发你

你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是pantene0807@163.com

xiaoliangstd, 2024-06-17:
> > 你好，请问你有邮箱嘛，我把这个包发你
> 
> 你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是[pantene0807@163.com](mailto:pantene0807@163.com)

啊？不好意思，我记得我们技术支持工程师拉过一个微信群，我已经发群里了，我以为他就是您。我发您下

Panting007, 2024-06-18:
> > > 你好，请问你有邮箱嘛，我把这个包发你
> > 
> > 
> > 你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是[pantene0807@163.com](mailto:pantene0807@163.com)
> 
> 啊？不好意思，我记得我们技术支持工程师拉过一个微信群，我已经发群里了，我以为他就是您。我发您下

好的 请问可以再次发一下之前的群聊二维码吗，还有一些其他问题想和大家一起交流一下

114554adaw, 2024-10-28:
您好可以给我也发一份吗？我也找不到但需要用11.5.2节中unitree_pointCloud_publisher的程序包， @xiaoliangstd ，我的邮箱是948026333@qq.com，谢谢@Panting007

mrbbang, 2025-06-13:
> > > > 你好，请问你有邮箱嘛，我把这个包发你
> > > 
> > > 
> > > 你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是[pantene0807@163.com](mailto:pantene0807@163.com)
> > 
> > 
> > 啊？不好意思，我记得我们技术支持工程师拉过一个微信群，我已经发群里了，我以为他就是您。我发您下
> 
> 好的 请问可以再次发一下之前的群聊二维码吗，还有一些其他问题想和大家一起交流一下

你好可以给我分享一份吗，我很需要谢谢

mrbbang, 2025-06-13:
> > > 你好，请问你有邮箱嘛，我把这个包发你
> > 
> > 
> > 你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是[pantene0807@163.com](mailto:pantene0807@163.com)
> 
> 啊？不好意思，我记得我们技术支持工程师拉过一个微信群，我已经发群里了，我以为他就是您。我发您下

你好可以给我分享一份吗，我也很需要谢谢

qin1-detl, 2025-07-06:
你好可以给我分享一份吗，我也很需要谢谢，我的邮箱是qin2003121_0825@qq.com

twistzzj, 2025-09-24:
> > > 你好，请问你有邮箱嘛，我把这个包发你
> > 
> > 
> > 你好，始终没有收到您关于unitree_pointCloud_publisher程序包的回复，我的邮箱是[pantene0807@163.com](mailto:pantene0807@163.com)
> 
> 啊？不好意思，我记得我们技术支持工程师拉过一个微信群，我已经发群里了，我以为他就是您。我发您下

你好，这个包我也需要，可以给我发一份吗，谢谢，我的邮箱是1063202215@qq.com