unitreerobotics/unitree_guide #76: compatibility with go2 model

AlessiaLiNoce, 2024-05-20:
Hi, is this unitree_guide also compatible for the go2 model?

xiaoliangstd, 2024-05-22:
Hello, currently there is no content related to the Go2 in this repository. We have a private unitree_guide package adapted for the Go2, specifically for the communication part. If you need it, I can send it to you.

AlessiaLiNoce, 2024-05-22:
If it's possible, yes, thank you.

AlessiaLiNoce, 2024-05-25:
> If it's possible, yes, thank you.

My email is alessia.linoce@gmail.com.  Also, is there some example on official  repository related to force control ?

Maan0798, 2024-06-11:
> Hello, currently there is no content related to the Go2 in this repository. We have a private unitree_guide package adapted for the Go2, specifically for the communication part. If you need it, I can send it to you.

Could you please send me a copy of this package as well? My email is fu_huangan@163.com. Thank you very much!

AlessiaLiNoce, 2024-06-21:
> Hello, currently there is no content related to the Go2 in this repository. We have a private unitree_guide package adapted for the Go2, specifically for the communication part. If you need it, I can send it to you.

Hi, is it still possible to have it?  Also, is there a specific example on gazebo in which I can try pure torque control on go2 ?

thisismoiself, 2024-07-01:
> Hello, currently there is no content related to the Go2 in this repository. We have a private unitree_guide package adapted for the Go2, specifically for the communication part. If you need it, I can send it to you.

Hello @xiaoliangstd , if you could also send it to me, i would be very grateful! My email is maxim.ritter@tum.de

Thank you!

closed 2024-07-01 by AlessiaLiNoce