unitreerobotics/unitree_guide #77: junior_ctrl文件在哪里

MGZD32, 2024-05-24:
[image]
找不到junior_ctrl

Git-BoHu, 2024-05-29:
我是这个报错：`./devel/lib/unitree_guide/junior_ctrl: error while loading shared libraries: libxmlrpcpp.so: cannot open shared object file: No such file or directory`

Silverster98, 2024-05-29:
> 我是这个报错：`./devel/lib/unitree_guide/junior_ctrl: error while loading shared libraries: libxmlrpcpp.so: cannot open shared object file: No such file or directory`

我的刚刚解决，我是把 `/opt/ros/noetic/lib` 这个路径 加到 `/etc/ld.so.conf` 文件里，然后 `ldconfig` 刷一下就好了。

xiaoliangstd, 2024-06-13:
> ![2024-05-24 12-15-33屏幕截图](https://private-user-images.githubusercontent.com/166226090/333442999-ccba2196-0973-4dce-900a-007dd2a591d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNjEyMDQsIm5iZiI6MTcxODI2MDkwNCwicGF0aCI6Ii8xNjYyMjYwOTAvMzMzNDQyOTk5LWNjYmEyMTk2LTA5NzMtNGRjZS05MDBhLTAwN2RkMmE1OTFkMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxM1QwNjQxNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZDRmODA5M2M5NTMxNDBkOTM0ZDI0MzA4YjIyY2NkMDg0YjQ2NGFkNmZhODcyMzRhNDdmOWE3NWM4OWM2ODExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UpXNVvSb72-9bh_Syk-Jj7oxib1HGuklg0Ma2vvdNGQ) 找不到junior_ctrl

是否是没有 catkin_make 进行编译呢

xiaoliangstd, 2024-06-13:
> > > 我是这个报错：`./devel/lib/unitree_guide/junior_ctrl: error while loading shared libraries: libxmlrpcpp.so: cannot open shared object file: No such file or directory`
> > 
> > 
> > 我的刚刚解决，我是把 `/opt/ros/noetic/lib` 这个路径 加到 `/etc/ld.so.conf` 文件里，然后 `ldconfig` 刷一下就好了。
> 
> 您提供的方法有效，问题已经解决了，谢谢！

https://support.unitree.com/home/zh/Algorithm_Practice/faq，这个文档中心有提及此问题