unitreerobotics/unitree_guide #79: WaveStatus中的SWING_ALL有什么用？同时在State_Trotting.cpp文件中的exit()函数中，为什么要设置_ctrlComp->setAllSwing()？

xliu0105, 2024-06-12:
我不太明白，在现有的代码及实现的功能中，SWING_ALL这个WaveStatus有什么用，这个应该是代表所有的足端都悬空吧，也许如果之后实现后空翻可能会用到这种状态？现有的代码这个有什么用。

在State_Trotting.cpp文件中的exit()函数的最后一行，代码设置了：_ctrlComp->setAllSwing()，请问这里是为什么？经过我测试，有这一行代码和没有这一行代码其实效果完全一样，而且我认为这一行代码其实是没啥用的。

xliu0105, 2024-06-13:
您好，感谢您的回答，但我还是有一些不理解。
在FSM::run()函数中，如果_mode == FSMMode::CHANGE，这时就会进行state切换，以从State_Trotting切换到State_FixedStand为例，首先运行State_Trotting的exit()函数，在这个exit()函数中最后一行执行了_ctrlComp->setAllSwing()；之后在FSM::run中继续执行_currentState = _nextState，即将State_FixedStand设为当前的State，然后执行State_FixedStand的enter()函数，在这个enter()函数中最后一行执行了_ctrlComp->setAllStance()。
在整个过程中，并没有执行runWaveGen()和estimator->run()啊，WaveStatus就已经换到STANCE_ALL了，所以我还是感觉这里那一行代码是多余的。

xiaoliangstd, 2024-06-13:
> 您好，感谢您的回答，但我还是有一些不理解。 在FSM::run()函数中，如果_mode == [...]

听你这么一说确实是的，但我认为这里之所以加上是因为，万一在 别的状态机里没有或者忘了设置，在 State_Trotting 里事先设置好，有一份保险。但从理性上看，确实是没有机会执行。

xiaoliangstd, 2024-06-13:
> 您好，感谢您的回答，但我还是有一些不理解。 在FSM::run()函数中，如果_mode == [...]

那您这里可以直接去掉也行，应该问题不大