unitreerobotics/unitree_guide #81: Catkin_make报错

Kane0114, 2024-07-19:
如图，按照教程克隆安装包以及安装了lcm之后，在catkin_ws中catkin_make报错，显示跳过不兼容的/home/kane/catkin_ws/src/unitree_guide/unitree_actuator_sdk/unitree_motor_ctrl/lib/libUnitree_motor_SDK_Linux.so when searching for -lUnitree_motor_SDK_Linux
[image]

[image]

Kane0114, 2024-07-19:
已解决，电脑arm64架构的，把unitree_actuatuor_sdk包删除就行了？不够删除之后对功能有影响吗？

xiaoliangstd, 2024-07-19:
> 已解决，电脑arm64架构的，把unitree_actuatuor_sdk包删除就行了？不够删除之后对功能有影响吗？

你好，这个unitree_actuator_sdk 仅仅是书本第二章内容的功能包，删除它，对学习主要的章节内容没有影响。