unitreerobotics/unitree_guide #84: 关于利用unitree_guide来模拟B1

Mars995, 2024-08-03:
你好，我想在Gazebo里模拟B1，我按照Go1和A1设定了相应类和参数，比如质量，惯性张量，腿长等参数，Xacro文件我目前用的是之前官方给我的Truck是18.276kg，质心在中心的版本，官网之前有25kg和现在29kg的。
但是我发现它并不能直接调用junior_ctrl控制，FixedStand状态无法站立，Trotting状态出现如下图这个情况。
[image]
所以问题一是想请问一下这个情况是什么原因导致的，看起来是controller失效了，但并没有触发安全保护，即进入Passiv状态。

我看完官网的书和unitree_ros文件后，发现问题出在Joint_controller_tool.cpp文件，好在三周前它更新了这部分代码，之前提供的lib.so似乎限制了它最大力矩在55,，但现在我使用这个仍然有问题，公式是calcTorque = posStiffness*(targetPos-currentPos) + velStiffness*(targetVel-currentVel) + targetTorque。对于FixedStand状态，targetTorque为0，它只提供了角度和速度变化，我们设定了PD值，但是Go1和A1只有13kg，而B1有50kg，这导致我需要把PD值调大4-5倍才能站立，虽然这个PD值不影响Trotting。所以问题二是想请问我是否需要给定一个初始力矩，还是说就应该这么设定PD值。

此外对于Trotting，有_ddPcd = _Kpp * _posError + _Kdp * _velError，我是否仍然需要调大Kpp和Kdp,但我的导师根据实验室里真实的B1机器人给我建议的PD参数是200和5这意味我使用的PD参数可能与真实不符。目前我使用的参数还是根据Go1，当我调大FixedStand的PD，再切换到Trotting它可以利用setStableGain的PD值维持站立，但一旦动起来，它又会因为力矩不够趴下去并且导致上图的错误。您是否能给我一些建议，或者我可以利用别的方法例如MPC解决这个问题？感谢您的回复。

xiaoliangstd, 2024-08-03:
问题一：我认为是 b1 机器人 并不适合用 单刚体模型控制和 非常简单的qp 控制器控制。或者需要尝试使用 MPC 控制器，也尝试使用 ocs2等来控制。   问题2：同上，因为这个控制器是非常简单的，B1机器人的腿部惯量还是太大了 ，这时候还使用单刚体模型，同时又是非常简单的qp,显然控制效果不行。

Mars995, 2024-08-03:
非常感谢您的回答，我会在之后尝试别的控制器

xiaoliangstd, 2024-08-03:
客气了

closed 2024-08-03 by xiaoliangstd