unitreerobotics/unitree_guide #85: 运行roslaunch unitree_move_base rvizMoveBase.launch 后出现问题

wxyanxw, 2024-08-08:
运行roslaunch unitree_move_base rvizMoveBase.launch 后，出现了这样的问题：
[image]
模型显示不全，且rviz界面报错，命令行也报错：[ WARN] [1723081750.374851194, 1739.182000000]: Timed out waiting for transform from base to odom to become available before running costmap, tf error: canTransform: target_frame odom does not exist.. canTransform returned after 1739.18 timeout was 0.1.
[ WARN] [1723081755.486179233, 1744.283000000]: Timed out waiting for transform from base to odom to become available before running costmap, tf error: canTransform: target_frame odom does not exist.. canTransform returned after 0.1 timeout was 0.1.

输入rosrun tf tf_echo base odom后出现：
Failure at 1933.471000000
Exception thrown:"odom" passed to lookupTransform argument source_frame does not exist. 
The current list of frames is:
Frame FL_calf_rotor exists with parent FL_thigh.
Frame FL_thigh exists with parent FL_hip.
Frame FL_foot exists with parent FL_calf.
Frame FL_calf exists with parent FL_thigh.
Frame FL_hip_rotor exists with parent trunk.
Frame trunk exists with parent base.
Frame FL_thigh_rotor exists with parent FL_hip.
Frame FL_hip exists with parent trunk.
Frame FR_calf_rotor exists with parent FR_thigh.
Frame FR_thigh exists with parent FR_hip.
Frame FR_foot exists with parent FR_calf.
Frame FR_calf exists with parent FR_thigh.
Frame FR_hip_rotor exists with parent trunk.
Frame FR_thigh_rotor exists with parent FR_hip.
Frame FR_hip exists with parent trunk.
Frame RL_calf_rotor exists with parent RL_thigh.
Frame RL_thigh exists with parent RL_hip.
Frame RL_foot exists with parent RL_calf.
Frame RL_calf exists with parent RL_thigh.
Frame RL_hip_rotor exists with parent trunk.
Frame RL_thigh_rotor exists with parent RL_hip.
Frame RL_hip exists with parent trunk.
Frame RR_calf_rotor exists with parent RR_thigh.
Frame RR_thigh exists with parent RR_hip.
Frame RR_foot exists with parent RR_calf.
Frame RR_calf exists with parent RR_thigh.
Frame RR_hip_rotor exists with parent trunk.
Frame RR_thigh_rotor exists with parent RR_hip.
Frame RR_hip exists with parent trunk.
Frame camera_chin exists with parent trunk.
Frame camera_face exists with parent trunk.
Frame camera_left exists with parent trunk.
Frame camera_rearDown exists with parent trunk.
Frame camera_right exists with parent trunk.
Frame camera_laserscan_link_left exists with parent camera_left.
Frame camera_laserscan_link_right exists with parent camera_right.
Frame camera_optical_chin exists with parent camera_chin.
Frame camera_optical_face exists with parent camera_face.
Frame camera_optical_left exists with parent camera_left.
Frame camera_optical_rearDown exists with parent camera_rearDown.
Frame camera_optical_right exists with parent camera_right.
Frame imu_link exists with parent trunk.
Frame ultraSound_face exists with parent trunk.
Frame ultraSound_left exists with parent trunk.
Frame ultraSound_right exists with parent trunk.

感觉可能是tf发布有问题？求各位大佬解答

wxyanxw, 2024-08-12:
后来将rviz中的fixed frame和reference frame改成base后解决，但不知道会不会影响后续操作
[image]

northpoleforce, 2024-08-12:
我也遇到这个问题，
1. 我的rviz帧率很低，而且一直警告（佬的机器是什么配置）
2. 发布目标位姿后机器狗也不动

佬是怎么解决的，教教我（可怜）
[image]
[image]

wxyanxw, 2024-08-12:
> 我也遇到这个问题，
>
>     1. 我的rviz帧率很低，而且一直警告（佬的机器是什么配置）
>
>     2. 发布目标位姿后机器狗也不动
>
>
> 佬是怎么解决的，教教我（可怜） [...]

别大佬了，我之前误导你了，别这样改，改回去!然后打开unitree_guide/CMakeList.txt，将前面的设置项改为set(MOVE_BASE ON)，这样CMakeList.txt就能定义COMPILE_WITH_MOVE_BASE宏，使得源文件中对应的代码得到编译。这句话是书本的原话，我犯了愚蠢的错误，没有看到导航的前置配置。

wxyanxw, 2024-08-12:
问题已解决，请看上面的回复！

northpoleforce, 2024-08-12:
感谢，顺带想问下同时运行gazebo和rviz，你的gazebo和rviz的帧率是多少 
我的gazebo 25fps，rviz 6fps
整个电脑卡卡的，你会有这个问题吗？如何解决的？

wxyanxw, 2024-08-13:
> 问题已解决，请看上面的回复！

我并没有这个问题，我的处理器是i5 11代，可能需要比较好的设备

zk0919, 2025-02-13:
> 问题已解决，请看上面的回复！

请问一下， 我的依旧不行，这是为什么

[image]

Nat-J, 2026-03-05:
如果gazebo非常非常卡，英伟达的卡的话，尝试运行`nvidia-smi`确认一下有没有装显卡驱动，没装驱动的话默认用cpu的核显，性能很弱，装完驱动后重启一下就好